syssi/esphome-votronic #28: Votronic Smart Shunt 400 S (unhandled message data)

Esylias, 2023-03-28:
[17:27:44][W][votronic:147]: Please provide the following unhandled message data (0xDA): AA.DA.00.00.00.00.F8.11.5E.07.00.00.2F.04.02.43 (16)

syssi, 2023-03-28:
```
     0  0xAA  Start of frame
     1  0xDA  Frame type
     2  0x00 0x00
     4  0x00 0x00
     6  0xF8 0x11: 0x11F8 = 4600
     8  0x5E 0x07: 0x075E = 1886
    10  0x00 0x00
    12  0x2F 0x04: 0x042F = 1071
    14  0x02
    15  0x43 CRC
```

Esylias, 2023-03-28:
Wow, fast Response 👍 
How can i help to figure out the problem?

Should i set some debug settings in esphome and post some Logs?

syssi, 2023-03-28:
Some logs would be nice. Are you able to receive and decode the traffic in general? Is this the only unsupported frame of your setup? Which devices do you own?

Esylias, 2023-03-28:
Yes im able to receive and decoder the traffic frrom my device "Votronic Smart Shunt 400 S"

The normal esphome debug ist showing only the 0xDA as unsupported frame

Esylias, 2023-03-28:
The Display LCD-Battery-Computer S will showing 

Battery Ah
Charging
Remainig Time 
Percent
Battery Status

Here ist the product PDF link https://www.votronic.de/images/Manuals/08.%20Measuring%20Devices%20and%20Displays/1263_1266_1269_e.pdf

Here ist the full product Link https://www.votronic.de/index.php/en/products/measuring-display-systems/battery-computer#downloads

syssi, 2023-03-28:
What's your battery capacity and the current state of charge (%)?

Esylias, 2023-03-28:
battery capacity
6  0xF8 0x11: 0x11F8 = 4600 = 460AH

charge
100%

syssi, 2023-03-28:
Is the battery voltage around 18.86V?

Esylias, 2023-03-28:
should be 12,9 - 13.41V

syssi, 2023-03-28:
I've pushed a feature branch to decode the new frame. Please update your YAML to:

```
substitutions:
  ...
  external_components_source: github://syssi/esphome-votronic@decode-battery-computer-info2
```

and flash you ESP again. You should see some additional log message:

https://github.com/syssi/esphome-votronic/pull/29/files

Esylias, 2023-03-28:
Before Update

[22:31:07][W][votronic:147]: Please provide the following unhandled message data (0xFA): AA.FA.2F.00.00.00.D2.02.00.0B.00.00.28.D0.00.F6 (16)

after update
[22:32:17][W][votronic:148]: Please provide the following unhandled message data (0xFA): AA.FA.2F.00.00.00.D3.02.00.09.00.00.28.D0.00.F5 (16)

syssi, 2023-03-29:
This looks like a bunch of new frames. I've updated the feature branch. Are you sure you did switch the branch by updating the `external_components` section? The output should look like this now:

```
[08:31:54][I][votronic:279]: Battery computer info2 data received
[08:31:54][VV][votronic:280]:   AA.DA.00.00.00.00.F8.11.5E.05.00.00.2F.04.02.41 (16)
[08:31:54][I][votronic:286]: Byte   2-3: 0x00 0x00 / 0 0 / 0
[08:31:54][I][votronic:288]: Byte   4-5: 0x00 0x00 / 0 0 / 0
[08:31:54][I][votronic:290]: Battery capacity: 460.0 Ah
[08:31:54][I][votronic:292]: Byte   8-9: 0x5E 0x05 / 94 5 / 1374
[08:31:54][I][votronic:294]: Byte 10-11: 0x00 0x00 / 0 0 / 0
[08:31:54][I][votronic:297]: Byte 12-13: 0x2F 0x04 / 47 4 / 1071
[08:31:54][I][votronic:300]: Byte    14: 0x02 / 2
[08:31:57][I][votronic:247]: Battery computer info1 data received
[08:31:57][VV][votronic:248]:   AA.CA.03.05.0F.05.C7.01.20.00.63.00.7B.FE.FF.39 (16)
[08:31:57][I][votronic:254]: Byte   2-3: 0x03 0x05 / 12.83 V
[08:31:57][I][votronic:256]: Byte   4-5: 0x0F 0x05 / 12.95 V
[08:31:57][I][votronic:258]: Byte   6-7: 0xC7 0x01 / 199 1 / 455
[08:31:57][I][votronic:260]: Byte   8-9: 0x20 0x00 / 32 0 / 32
[08:31:57][I][votronic:262]: Sate of charge?: 0x63 0x00 / 99 0 / 99 %
[08:31:57][I][votronic:264]: Byte 12-13: 0x7B 0xFE / 123 254 / 65147
[08:31:57][I][votronic:267]: Byte    14: 0xFF / 255
[08:31:59][I][votronic:312]: Unnamed data received
[08:31:59][VV][votronic:313]:   AA.FA.2F.00.00.00.D2.02.00.0A.00.00.28.D0.00.F7 (16)
[08:31:59][I][votronic:319]: Byte   2-3: 0x2F 0x00 / 47 0 / 47
[08:31:59][I][votronic:321]: Byte   4-5: 0x00 0x00 / 0 0 / 0
[08:31:59][I][votronic:323]: Byte   6-7: 0xD2 0x02 / 210 2 / 722
[08:31:59][I][votronic:325]: Byte   8-9: 0x00 0x0A / 0 10 / 2560
[08:31:59][I][votronic:327]: Byte 10-11: 0x00 0x00 / 0 0 / 0
[08:31:59][I][votronic:330]: Byte 12-13: 0x28 0xD0 / 40 208 / 53288
[08:31:59][I][votronic:333]: Byte    14: 0x00 / 0
```

Esylias, 2023-03-29:
[11:08:30][VV][votronic:108]: RX <- AA.FA.2F.00.50.05.01.03.00.F8.FF.FF.28.05.00.57 (16)
[11:08:30][D][uart_debug:114]: <<< AA:FA:2F:00:50:05:01:03:00:F8:FF:FF:28:05:00:57
[11:08:30][VV][votronic:108]: RX <- AA.CA.56.05.25.05.C3.01.20.00.62.00.B0.01.00.88 (16)
[11:08:31][D][uart_debug:114]: <<< AA:CA:56:05:25:05:C3:01:20:00:62:00:B0:01:00:88
[11:08:31][VV][votronic:108]: RX <- AA.DA.00.00.00.00.F8.11.5E.02.00.00.2F.04.02.46 (16)
[11:08:31][I][votronic:243]: Battery computer info2 data received
[11:08:31][VV][votronic:244]:   AA.DA.00.00.00.00.F8.11.5E.02.00.00.2F.04.02.46 (16)
[11:08:31][I][votronic:250]: Byte   2-3: 0x00 0x00 / 0 0 / 0
[11:08:31][I][votronic:252]: Byte   4-5: 0x00 0x00 / 0 0 / 0
[11:08:31][I][votronic:254]: Battery capacity: 460.0 Ah
[11:08:32][I][votronic:256]: Byte   8-9: 0x5E 0x02 / 94 2 / 606
[11:08:32][I][votronic:259]: Byte 10-11: 0x00 0x00 / 0 0 / 0
[11:08:32][I][votronic:262]: Byte 12-13: 0x2F 0x04 / 47 4 / 1071
[11:08:32][I][votronic:264]: Byte    14: 0x02 / 2

My Display is currently showing

Charging Status
98% Battery Status
451 Ah
"Off"
Charging with 0.4A
Board battery 13,7V
Start battery 13,2 V
Remaining time -.-h (Cause display need some time to calculate)

syssi, 2023-03-29:
Could you update your ESP another time? You should see the log of three different frames at the moment:

```
[08:31:54][I][votronic:279]: Battery computer info2 data received
...
[08:31:57][I][votronic:247]: Battery computer info1 data received
...
[08:31:59][I][votronic:312]: Unnamed data received
...
```

The state of charge f.e. is part of the second `info1` frame IMO.

syssi, 2023-03-29:
Could you set the `throttle` parameter to `0s` and try again? This time all three frames should be decoded.

syssi, 2023-03-29:
Could you provide some more display values plus the three status frames for comparison?

Esylias, 2023-03-29:
what u

> [11:08:30][VV][votronic:108]: RX <- AA.FA.2F.00.50.05.01.03.00.F8.FF.FF.28.05.00.57 (16) [11:08:30][D][uart_debug:114]: <<< AA:FA:2F:00:50:05:01:03:00:F8:FF:FF:28:05:00:57 [...]

these are the display infos, what u mean with "three status frames"?

this is what i get all the time

[16:43:22][D][uart_debug:114]: <<< AA:DA:00:00:00:00:F8:11:5E:09:00:00:2F:04:02:4D
[16:43:23][VV][votronic:108]: RX <- AA.FA.2F.00.50.05.00.03.00.FE.FF.FF.28.05.00.50 (16)
**[16:43:23][I][votronic:313]: Unnamed data received**
[16:43:23][VV][votronic:314]:   AA.FA.2F.00.50.05.00.03.00.FE.FF.FF.28.05.00.50 (16)
[16:43:23][I][votronic:320]: Byte   2-3: 0x2F 0x00 / 47 0 / 47
[16:43:23][I][votronic:322]: Byte   4-5: 0x50 0x05 / 80 5 / 1360
[16:43:23][I][votronic:324]: Byte   6-7: 0x00 0x03 / 0 3 / 768
[16:43:23][I][votronic:326]: Byte   8-9: 0x00 0xFE / 0 254 / 65024
[16:43:23][I][votronic:329]: Byte 10-11: 0xFF 0xFF / 255 255 / 65535
[16:43:23][I][votronic:332]: Byte 12-13: 0x28 0x05 / 40 5 / 1320
[16:43:23][I][votronic:334]: Byte    14: 0x00 / 0

[16:43:23][D][uart_debug:114]: <<< AA:FA:2F:00:50:05:00:03:00:FE:FF:FF:28:05:00:50
[16:43:23][VV][votronic:108]: RX <- AA.CA.55.05.23.05.C5.01.20.00.63.00.2B.00.00.10 (16)
**[16:43:23][I][votronic:247]: Battery computer info1 data received**
[16:43:23][VV][votronic:248]:   AA.CA.55.05.23.05.C5.01.20.00.63.00.2B.00.00.10 (16)
[16:43:23][I][votronic:254]: Battery voltage: 13.65 V
[16:43:23][I][votronic:256]: Second battery voltage: 13.15 V
[16:43:23][I][votronic:258]: Byte   6-7: 0xC5 0x01 / 197 1 / 453
[16:43:23][I][votronic:260]: Byte   8-9: 0x20 0x00 / 32 0 / 32
[16:43:23][I][votronic:263]: Sate of charge?: 0x63 0x00 / 99 0 / 99 %
[16:43:23][I][votronic:266]: Byte 12-13: 0x2B 0x00 / 43 0 / 43
[16:43:23][I][votronic:268]: Byte    14: 0x00 / 0

[16:43:23][D][uart_debug:114]: <<< AA:CA:55:05:23:05:C5:01:20:00:63:00:2B:00:00:10
[16:43:24][VV][votronic:108]: RX <- AA.DA.00.00.00.00.F8.11.5E.09.00.00.2F.04.02.4D (16)
**[16:43:24][I][votronic:280]: Battery computer info2 data received**
[16:43:24][VV][votronic:281]:   AA.DA.00.00.00.00.F8.11.5E.09.00.00.2F.04.02.4D (16)
[16:43:24][I][votronic:287]: Byte   2-3: 0x00 0x00 / 0 0 / 0
[16:43:24][I][votronic:289]: Byte   4-5: 0x00 0x00 / 0 0 / 0
[16:43:24][I][votronic:291]: Battery capacity: 460.0 Ah
[16:43:24][I][votronic:293]: Byte   8-9: 0x5E 0x09 / 94 9 / 2398
[16:43:24][I][votronic:296]: Byte 10-11: 0x00 0x00 / 0 0 / 0
[16:43:24][I][votronic:299]: Byte 12-13: 0x2F 0x04 / 47 4 / 1071
[16:43:24][I][votronic:301]: Byte    14: 0x02 / 2

syssi, 2023-03-29:
I tried to identify some more bytes. Please update your ESP and provide another log. Could you create some load (discharge the battery) to make sure the bytes holding the `current` are identified correctly?

Esylias, 2023-03-29:
[20:51:25][D][uart_debug:114]: <<< AA:DA:00:00:00:00:F8:11:5E:09:00:00:2F:04:02:4D
[20:51:26][VV][votronic:108]: RX <- AA.FA.2F.00.00.00.EB.02.00.0C.00.00.28.D0.00.C8 (16)
[20:51:26][I][votronic:313]: Unnamed data received
[20:51:26][VV][votronic:314]:   AA.FA.2F.00.00.00.EB.02.00.0C.00.00.28.D0.00.C8 (16)
[20:51:26][I][votronic:320]: Byte   2-3: 0x2F 0x00 / 47 0 / 47
[20:51:26][I][votronic:322]: Byte   4-5: 0x00 0x00 / 0 0 / 0
[20:51:26][I][votronic:324]: Byte   6-7: 0xEB 0x02 / 235 2 / 747
[20:51:26][I][votronic:326]: Byte   8-9: 0x00 0x0C / 0 12 / 3072
[20:51:26][I][votronic:329]: Byte 10-11: 0x00 0x00 / 0 0 / 0
[20:51:26][I][votronic:332]: Byte 12-13: 0x28 0xD0 / 40 208 / 53288
[20:51:26][I][votronic:334]: Byte    14: 0x00 / 0

[20:51:26][D][uart_debug:114]: <<< AA:FA:2F:00:00:00:EB:02:00:0C:00:00:28:D0:00:C8
[20:51:26][VV][votronic:108]: RX <- AA.CA.2F.05.06.05.CC.01.20.00.64.00.F9.FD.FF.91 (16)
[20:51:26][I][votronic:247]: Battery computer info1 data received
[20:51:26][VV][votronic:248]:   AA.CA.2F.05.06.05.CC.01.20.00.64.00.F9.FD.FF.91 (16)
[20:51:26][I][votronic:254]: Battery voltage: 13.27 V
[20:51:26][I][votronic:256]: Second battery voltage: 12.86 V
[20:51:26][I][votronic:258]: Byte   6-7: 0xCC 0x01 / 204 1 / 460
[20:51:26][I][votronic:260]: Byte   8-9: 0x20 0x00 / 32 0 / 32
[20:51:26][I][votronic:263]: Sate of charge?: 0x64 0x00 / 100 0 / 100 %
[20:51:26][I][votronic:266]: Byte 12-13: 0xF9 0xFD / 249 253 / 65017
[20:51:26][I][votronic:268]: Byte    14: 0xFF / 255

[20:51:26][D][uart_debug:114]: <<< AA:CA:2F:05:06:05:CC:01:20:00:64:00:F9:FD:FF:91
[20:51:27][VV][votronic:108]: RX <- AA.DA.00.00.00.00.F8.11.5E.09.00.00.2F.04.02.4D (16)
[20:51:27][I][votronic:280]: Battery computer info2 data received
[20:51:27][VV][votronic:281]:   AA.DA.00.00.00.00.F8.11.5E.09.00.00.2F.04.02.4D (16)
[20:51:27][I][votronic:287]: Byte   2-3: 0x00 0x00 / 0 0 / 0
[20:51:27][I][votronic:289]: Byte   4-5: 0x00 0x00 / 0 0 / 0
[20:51:27][I][votronic:291]: Battery capacity: 460.0 Ah
[20:51:27][I][votronic:293]: Byte   8-9: 0x5E 0x09 / 94 9 / 2398
[20:51:27][I][votronic:296]: Byte 10-11: 0x00 0x00 / 0 0 / 0
[20:51:27][I][votronic:299]: Byte 12-13: 0x2F 0x04 / 47 4 / 1071
[20:51:27][I][votronic:301]: Byte    14: 0x02 / 2

syssi, 2023-03-29:
You are not using the most recent version of the feature branch. Please make sure the `refresh` interval is set to `0s` + rebuild and flash your ESP again:

```
external_components:
  - source: github://syssi/esphome-votronic@decode-battery-computer-info2
    refresh: 0s
```

Esylias, 2023-03-29:
Can we communicate everywhere directly? Skype, Discord Chat or something? Maybe direclty is easier

syssi, 2023-03-29:
[image]

Could you provide the following settings (values) of your setup:

- Battery Type
- Terminal off threshold (voltage)
- Terminal on threshold (capacity)
- Terminal off threshold (capacity)
- Memory lowest voltage
- Memory highest voltage
- Memory lowest capacity
- Memory highest capacity
- Hours ago the battery got fully charged

> Can we communicate everywhere directly? Skype, Discord Chat or something? Maybe directly is easier

Discord: syssi#5175

Esylias, 2023-03-29:
[20:58:39][D][uart_debug:114]: <<< AA:FA:2F:00:00:00:E6:02:00:0B:00:00:28:D0:00:C2
[20:58:40][VV][votronic:108]: RX <- AA.CA.26.05.03.05.CB.01.20.00.64.00.25.FE.FF.45 (16)
[20:58:40][I][votronic:260]: Battery computer info1 data received
[20:58:40][VV][votronic:261]: i1  AA.CA.26.05.03.05.CB.01.20.00.64.00.25.FE.FF.45 (16)
[20:58:40][I][votronic:267]: i1 Battery voltage: 13.18 V
[20:58:40][D][sensor:127]: 'votronic-smart-shunt battery voltage': Sending state 13.18000 V with 2 decimals of accuracy
[20:58:40][I][votronic:271]: i1 Second battery voltage: 12.83 V
[20:58:40][D][sensor:127]: 'votronic-smart-shunt secondary battery voltage': Sending state 12.83000 V with 2 decimals of accuracy
[20:58:40][I][votronic:274]: i1 Capacity remaining: 459 Ah
[20:58:40][I][votronic:276]: i1 Byte   8-9: 0x20 0x00 / 32 0 / 32
[20:58:40][I][votronic:278]: i1 State of charge: 100 %
[20:58:40][D][sensor:127]: 'votronic-smart-shunt state of charge': Sending state 100.00000 % with 0 decimals of accuracy
[20:58:40][I][votronic:281]: i1 Current: -4.75 A
[20:58:40][D][sensor:127]: 'votronic-smart-shunt current': Sending state -4.75000 A with 3 decimals of accuracy
[20:58:40][D][sensor:127]: 'votronic-smart-shunt power': Sending state -62.60500 W with 2 decimals of accuracy
[20:58:40][I][votronic:289]: i1 Byte    14: 0xFF / 255
[20:58:40][D][uart_debug:114]: <<< AA:CA:26:05:03:05:CB:01:20:00:64:00:25:FE:FF:45
[20:58:41][VV][votronic:108]: RX <- AA.DA.00.00.00.00.F8.11.5E.09.00.00.2F.04.02.4D (16)
[20:58:41][I][votronic:307]: Battery computer info2 data received
[20:58:41][VV][votronic:308]: i2  AA.DA.00.00.00.00.F8.11.5E.09.00.00.2F.04.02.4D (16)
[20:58:41][I][votronic:314]: i2 Byte   2-3: 0x00 0x00 / 0 0 / 0
[20:58:41][I][votronic:316]: i2 Byte   4-5: 0x00 0x00 / 0 0 / 0
[20:58:41][I][votronic:318]: i2 Battery capacity: 460.0 Ah
[20:58:41][I][votronic:320]: i2 Byte   8-9: 0x5E 0x09 / 94 9 / 2398
[20:58:41][I][votronic:323]: i2 Byte 10-11: 0x00 0x00 / 0 0 / 0
[20:58:41][I][votronic:326]: i2 Byte 12-13: 0x2F 0x04 / 47 4 / 1071
[20:58:41][I][votronic:328]: i2 Byte    14: 0x02 / 2
[20:58:41][D][uart_debug:114]: <<< AA:DA:00:00:00:00:F8:11:5E:09:00:00:2F:04:02:4D
[20:58:42][VV][votronic:108]: RX <- AA.FA.2F.00.00.00.E6.02.00.0B.00.00.28.D0.00.C2 (16)
[20:58:42][I][votronic:346]: Battery computer info3 data received
[20:58:42][VV][votronic:347]: i3  AA.FA.2F.00.00.00.E6.02.00.0B.00.00.28.D0.00.C2 (16)
[20:58:42][I][votronic:353]: i3 Byte   2-3: 0x2F 0x00 / 47 0 / 47
[20:58:42][I][votronic:355]: i3 Byte   4-5: 0x00 0x00 / 0 0 / 0
[20:58:42][I][votronic:357]: i3 Byte   6-7: 0xE6 0x02 / 230 2 / 742
[20:58:42][I][votronic:359]: i3 Byte   8-9: 0x00 0x0B / 0 11 / 2816
[20:58:42][I][votronic:362]: i3 Byte 10-11: 0x00 0x00 / 0 0 / 0
[20:58:42][I][votronic:365]: i3 Byte 12-13: 0x28 0xD0 / 40 208 / 53288
[20:58:42][I][votronic:367]: i3 Byte    14: 0x00 / 0

Esylias, 2023-03-29:
**### Current display values**
Off
459 Ah
100%
802,5h
-0,4 / 0,5A 
Starter battery 12,7V
Second battery 13V

**### Current display settings**
460 Ah
typ 47
Off 10,5V
On 101%
Off 30%
0,4V (arrow down)
15V (arrow up)
0% (arrow down)
100% (arrow up)
0h
LED 100%

Esylias, 2023-03-29:
[21:30:11][D][uart_debug:114]: <<< AA:FA:2F:00:00:00:D9:02:00:0A:00:00:28:D0:00:FC
[21:30:12][VV][votronic:108]: RX <- AA.CA.0F.05.FC.04.CB.01.20.00.64.00.50.FE.FF.E7 (16)
[21:30:12][I][votronic:260]: Battery computer info1 data received
[21:30:12][VV][votronic:261]: i1  AA.CA.0F.05.FC.04.CB.01.20.00.64.00.50.FE.FF.E7 (16)
[21:30:12][I][votronic:267]: i1 Battery voltage: 12.95 V
[21:30:12][D][sensor:127]: 'votronic-smart-shunt battery voltage': Sending state 12.95000 V with 2 decimals of accuracy
[21:30:12][I][votronic:271]: i1 Second battery voltage: 12.76 V
[21:30:12][D][sensor:127]: 'votronic-smart-shunt secondary battery voltage': Sending state 12.76000 V with 2 decimals of accuracy
[21:30:12][I][votronic:274]: i1 Capacity remaining: 459 Ah
[21:30:12][I][votronic:276]: i1 Byte   8-9: 0x20 0x00 / 32 0 / 32
[21:30:12][I][votronic:278]: i1 State of charge: 100 %
[21:30:12][D][sensor:127]: 'votronic-smart-shunt state of charge': Sending state 100.00000 % with 0 decimals of accuracy
[21:30:12][I][votronic:281]: i1 Current: -4.32 A
[21:30:12][D][sensor:127]: 'votronic-smart-shunt current': Sending state -4.32000 A with 3 decimals of accuracy
[21:30:12][D][sensor:127]: 'votronic-smart-shunt power': Sending state -55.94400 W with 2 decimals of accuracy
[21:30:12][I][votronic:289]: i1 Byte    14: 0xFF / 255
[21:30:12][D][uart_debug:114]: <<< AA:CA:0F:05:FC:04:CB:01:20:00:64:00:50:FE:FF:E7
[21:30:12][VV][votronic:108]: RX <- AA.DA.00.00.00.00.F8.11.5E.09.00.00.2F.04.02.4D (16)
[21:30:12][I][votronic:307]: Battery computer info2 data received
[21:30:12][VV][votronic:308]: i2  AA.DA.00.00.00.00.F8.11.5E.09.00.00.2F.04.02.4D (16)
[21:30:12][I][votronic:314]: i2 Byte   2-3: 0x00 0x00 / 0 0 / 0
[21:30:12][I][votronic:316]: i2 Byte   4-5: 0x00 0x00 / 0 0 / 0
[21:30:13][I][votronic:318]: i2 Battery capacity: 460.0 Ah
[21:30:13][I][votronic:320]: i2 Byte   8-9: 0x5E 0x09 / 94 9 / 2398
[21:30:13][I][votronic:323]: i2 Byte 10-11: 0x00 0x00 / 0 0 / 0
[21:30:13][I][votronic:326]: i2 Byte 12-13: 0x2F 0x04 / 47 4 / 1071
[21:30:13][I][votronic:328]: i2 Byte    14: 0x02 / 2
[21:30:13][D][uart_debug:114]: <<< AA:DA:00:00:00:00:F8:11:5E:09:00:00:2F:04:02:4D
[21:30:13][VV][votronic:108]: RX <- AA.FA.2F.00.00.00.D9.02.00.0B.00.00.28.D0.00.FD (16)
[21:30:14][I][votronic:346]: Battery computer info3 data received
[21:30:14][VV][votronic:347]: i3  AA.FA.2F.00.00.00.D9.02.00.0B.00.00.28.D0.00.FD (16)
[21:30:14][I][votronic:353]: i3 Byte   2-3: 0x2F 0x00 / 47 0 / 47
[21:30:14][I][votronic:355]: i3 Byte   4-5: 0x00 0x00 / 0 0 / 0
[21:30:14][I][votronic:357]: i3 Byte   6-7: 0xD9 0x02 / 217 2 / 729
[21:30:14][I][votronic:359]: i3 Byte   8-9: 0x00 0x0B / 0 11 / 2816
[21:30:14][I][votronic:362]: i3 Byte 10-11: 0x00 0x00 / 0 0 / 0
[21:30:14][I][votronic:365]: i3 Byte 12-13: 0x28 0xD0 / 40 208 / 53288
[21:30:14][I][votronic:367]: i3 Byte    14: 0x00 / 0

syssi, 2023-03-29:
```
# Discharging
AA.CA.0F.05.FC.04.CB.01.20.00.64.00.50.FE.FF.E7 (16)
AA.DA.00.00.00.00.F8.11.5E.09.00.00.2F.04.02.4D (16)
AA.FA.2F.00.00.00.D9.02.00.0B.00.00.28.D0.00.FD (16)
```

```
# Charging
AA.CA.55.05.23.05.C5.01.20.00.63.00.2B.00.00.10 (16)
AA.DA.00.00.00.00.F8.11.5E.09.00.00.2F.04.02.4D (16)
AA.FA.2F.00.50.05.00.03.00.FE.FF.FF.28.05.00.50 (16)
```

syssi, 2023-04-14:
@Esylias Could you verify the battery voltages are labeled correctly? The primary battery is the "Bordbatterie" and the secondary battery is your starter battery.

closed 2023-04-21 by syssi