syssi/esphome-votronic #41: Problem with the  Battery Charger VBCS 60/40/430 Triple-CI  (3246)

whisker73, 2023-06-11:
Hello

first off all thanks for the nice implementation. But i have Problems with the Triple Charger
see the log , i get many off the Invalid Header messages and then the esp8266 reset. it seems the triple send many different messages and also one unknown:
[W][votronic:149]: Please provide the following unhandled message data (0x4A): AA.4A.00.00.00.00.00.00.00.00.00.00.00.00.00.4A (16)
[logs (4).txt](https://github.com/syssi/esphome-votronic/files/11714170/logs.4.txt)
but in the end is the reset the biggest problem

best regards

syssi, 2023-06-11:
Interesting. I will have a look! How long is your cable between the ESP and your votronic device?

whisker73, 2023-06-11:
about 2m, I've had it up until now via a splitter in parallel with the Jupiter display and a Raspberry Pi with Node Red and it works fine there. now i am change to home assistant and on the ESP it won't work well with the splitter and I have it right on the triple charger, I'll shorten the cable and try it again. for me looks like he lost the syncronisation and then you will get unbelivable amount of Invalid header messages. 
I mean it basically works, sometimes he loses the ESP because of the reset, ok so far.

syssi, 2023-06-11:
I've added some additional log statements to the receiver/parser: https://github.com/syssi/esphome-votronic/pull/42/files

Please update your YAML to use this feature branch, flash the ESP and provide another log. It should contain some useful details this time.

```
substitutions:
  external_components_source: github://syssi/esphome-votronic@debug-decoder
```

whisker73, 2023-06-11:
ok i have shorten die cable to 40cm and here is the log 
[logs_votronic_logs (1).txt](https://github.com/syssi/esphome-votronic/files/11714653/logs_votronic_logs.1.txt)

 thanks a lot :-D

syssi, 2023-06-11:
This RX buffer is filled with two bytes most of the time:

```
[14:54:56][W][votronic:099]: Invalid header: 0x16
[14:54:56][W][votronic:080]: Buffer cleared due to reset: 16.AA
[14:54:56][W][votronic:099]: Invalid header: 0x1A
[14:54:56][W][votronic:080]: Buffer cleared due to reset: 1A.68
[14:54:56][W][votronic:099]: Invalid header: 0x05
[14:54:56][W][votronic:080]: Buffer cleared due to reset: 05.75
[14:54:56][W][votronic:099]: Invalid header: 0x08
[14:54:56][W][votronic:080]: Buffer cleared due to reset: 08.00
[14:54:56][W][votronic:099]: Invalid header: 0x00
[14:54:56][W][votronic:080]: Buffer cleared due to reset: 00.00
[14:54:56][W][votronic:099]: Invalid header: 0xE0
[14:54:56][W][votronic:080]: Buffer cleared due to reset: E0.A6
[14:54:56][W][votronic:099]: Invalid header: 0x1F
[14:54:56][W][votronic:080]: Buffer cleared due to reset: 1F.56
[14:54:56][W][votronic:099]: Invalid header: 0x22
```

The receiver/parser tries to find the preamble at the first position of the buffer. If the buffer doesn't start with 0xAA it gets flushed/cleared. If the preamble is on the second position it gets eliminated and the full frame is ruined.

I've to think about it.

whisker73, 2023-06-11:
yeah in node red was easier the serial component has this inside, i am not so good in c++ because of that 
[image]

i try to understand your c++ code but i am not good in c++ , i have no idea how do you initialize the serial port or better here the software serial and where the rx_buffer will filled :-D

syssi, 2023-06-11:
Give me some time. I will provide a improved version as soon as possible!

whisker73, 2023-06-11:
no stress :-D 

thanks a lot !!!

syssi, 2023-06-12:
I was able to identify a nasty bug. I've merged a fix to `main` already. Please update your YAML back to the main branch and try again. The amount of `Invalid header` warnings and the unsupported frame type should be reduced now.

closed 2023-06-12 by syssi

whisker73, 2023-06-13:
Hi

I will check it today evening and send you a feedback

Thanks

2023, 18:09:

whisker73, 2023-06-13:
hi again

works better , between 10-->30 invalid header but much more correct data

thanks a lot

PR #42 (2023-06-12): Fix preamble synchronization
Fixes #41