syssi/esphome-votronic #43: Triple lader has more then one input with the same name but different values

whisker73, 2023-06-21:
hi 

First Problem is , how you can see below  the charger send two different frames like below :

both of them have 

 'votronic-solar-charger battery status': 
'votronic-solar-charger controller active': 
'votronic-solar-charger current reduction'
'votronic-solar-charger controller temperature'

but different values, means they toggle then in Homeassistant between this two different values.

i think one is for the solar and the other for b2b or landpower

is it possible to seperate this ?  

Second : The charger controller temperature from charger data is wrong 

thanks !!

[09:41:43][I][votronic:173]: **Solar charger data received**
[09:41:43][D][sensor:093]: 'votronic-solar-charger battery voltage': Sending state 13.84000 V with 2 decimals of accuracy
[09:41:43][D][sensor:093]: '**votronic-solar-charger controller temperature**': Sending state 31.00000 °C with 0 decimals of accuracy
[09:41:43][D][sensor:093]: 'votronic-solar-charger charging mode setting id': Sending state 86.00000  with 0 decimals of accuracy
[09:41:43][D][text_sensor:064]: 'votronic-solar-charger charging mode setting': Sending state 'LiFePo4 14.6V'
[09:41:43][D][sensor:093]: 'votronic-solar-charger battery status bitmask': Sending state 34.00000  with 0 decimals of accuracy
[09:41:43][D][text_sensor:064]: '**votronic-solar-charger battery status'**: Sending state **'U1 phase'**
[09:41:43][D][binary_sensor:036]: '**votronic-solar-charger controller active'**: Sending state **ON**
[09:41:44][D][binary_sensor:036]: '**votronic-solar-charger current reduction**': Sending state **ON**

[09:41:45][I][votronic:219]: **Charger data received**
[09:41:45][D][sensor:093]: 'votronic-solar-charger battery voltage': Sending state 13.84000 V with 2 decimals of accuracy
[09:41:45][D][sensor:093]: 'votronic-solar-charger secondary battery voltage': Sending state 12.46000 V with 2 decimals of accuracy
[09:41:45][D][sensor:093]: 'votronic-solar-charger current': Sending state 0.00000 A with 3 decimals of accuracy
[09:41:45][D][sensor:093]: 'votronic-solar-charger power': Sending state 0.00000 W with 2 decimals of accuracy
[09:41:45][D][sensor:093]: 'votronic-solar-charger state of charge': Sending state 0.00000 % with 0 decimals of accuracy
[09:41:45][D][sensor:093]: '**votronic-solar-charger controller temperature**': Sending state **166.00000 °C** with 0 decimals of accuracy
[09:41:45][D][sensor:093]: 'votronic-solar-charger charging mode setting id': Sending state 86.00000  with 0 decimals of accuracy
[09:41:45][D][text_sensor:064]: 'votronic-solar-charger charging mode setting': Sending state 'LiFePo4 14.6V'
[09:41:45][D][sensor:093]: 'votronic-solar-charger battery status bitmask': Sending state 40.00000  with 0 decimals of accuracy
[09:41:45][D][text_sensor:064]: '**votronic-solar-charger battery status**': Sending state '**U3 phase**'
[09:41:45][D][sensor:093]: 'votronic-solar-charger charging controller status bitmask': Sending state 0.00000  with 0 decimals of accuracy
[09:41:45][D][text_sensor:064]: 'votronic-solar-charger charging controller status': Sending state 'Standby'
[09:41:45][D][binary_sensor:036]: '**votronic-solar-charger controller active**': Sending state **OFF**
[09:41:45][D][binary_sensor:036]: '**votronic-solar-charger current reduction**': Sending state **OFF**

syssi, 2023-06-21:
Could you extend your YAML to write the RAW data to the log and provide a log capture?

https://github.com/syssi/esphome-votronic/blob/main/esp8266-charger-example-debug.yaml#L3-L27

This will help to get a better understanding which frames are available on your bus. Could you provide your YAML too? Did you merge different YAMLs because of the different Votronic devices?

whisker73, 2023-06-21:
of course 
[logs_votronic_run.txt](https://github.com/syssi/esphome-votronic/files/11817775/logs_votronic_run.txt)
[votronic.yaml.txt](https://github.com/syssi/esphome-votronic/files/11817809/votronic.yaml.txt)

syssi, 2023-06-21:
Traffic to reproduce the issue:

```
[...]
AA.1A.2F.05.C1.06.00.00.00.E0.A5.18.56.20.01.DD
AA.4A.00.00.00.00.00.00.00.00.00.00.00.00.00.4A
AA.3A.2E.05.D5.04.00.00.00.01.00.A3.56.28.00.1C
AA.7A.2E.05.D5.04.00.00.00.00.00.A3.56.28.00.5D

AA.1A.2F.05.C1.06.00.00.00.E0.A3.19.56.20.01.DA
AA.4A.00.00.00.00.00.00.00.00.00.00.00.00.00.4A
AA.3A.2F.05.D6.04.00.00.00.01.00.A2.56.28.00.1F
AA.7A.2F.05.D6.04.00.00.00.00.00.A2.56.28.00.5E

AA.1A.2F.05.C1.06.00.00.00.E0.A2.19.56.20.01.DB
AA.4A.00.00.00.00.00.00.00.00.00.00.00.00.00.4A
AA.3A.2F.05.D5.04.00.00.00.01.00.A2.56.28.00.1C
AA.7A.2F.05.D5.04.00.00.00.00.00.A2.56.28.00.5D

AA.1A.2F.05.C1.06.00.00.00.E0.A2.19.56.20.01.DB
AA.4A.00.00.00.00.00.00.00.00.00.00.00.00.00.4A
AA.3A.2F.05.D5.04.00.00.00.01.00.A2.56.28.00.1C
AA.7A.2F.05.D5.04.00.00.00.00.00.A2.56.28.00.5D

AA.1A.2F.05.BF.06.00.00.00.E0.A2.19.56.20.01.A5
AA.4A.00.00.00.00.00.00.00.00.00.00.00.00.00.4A
AA.3A.2F.05.D5.04.00.00.00.01.00.A2.56.28.00.1C
AA.7A.2F.05.D5.04.00.00.00.00.00.A2.56.28.00.5D

AA.1A.2F.05.BF.06.00.00.00.E0.A2.19.56.20.01.A5
AA.4A.00.00.00.00.00.00.00.00.00.00.00.00.00.4A
AA.3A.2F.05.D5.04.00.00.00.01.00.A2.56.28.00.1C
AA.7A.2F.05.D5.04.00.00.00.00.00.A2.56.28.00.5D

AA.1A.2F.05.BF.06.00.00.00.E0.A2.19.56.20.01.A5
AA.4A.00.00.00.00.00.00.00.00.00.00.00.00.00.4A
AA.3A.2F.05.D5.04.00.00.00.01.00.A2.56.28.00.1C
AA.7A.2F.05.D5.04.00.00.00.00.00.A2.56.28.00.5D

AA.1A.2F.05.BF.06.00.00.00.E0.A2.19.56.20.01.A5
AA.4A.00.00.00.00.00.00.00.00.00.00.00.00.00.4A
AA.3A.2F.05.D6.04.00.00.00.01.00.A2.56.28.00.1F
AA.7A.2F.05.D6.04.00.00.00.00.00.A2.56.28.00.5E

AA.1A.2F.05.BF.06.00.00.00.E0.A2.19.56.20.01.A5
AA.4A.00.00.00.00.00.00.00.00.00.00.00.00.00.4A
AA.3A.2F.05.D5.04.00.00.00.01.00.A2.56.28.00.1C
AA.7A.2F.05.D5.04.00.00.00.00.00.A2.56.28.00.5D
[...]
```

syssi, 2023-06-21:
I've introduced some new sensors. Please extend your YAML and try again: https://github.com/syssi/esphome-votronic/blob/main/esp8266-triple-charger-example.yaml

closed 2023-06-21 by syssi

whisker73, 2023-06-21:
thanks a lot i will check it , btw the triple charger needs 500-700ms rx_timeout  otherwise you will get a high amount of invalid headers because of the timeout buffer reset

br Holger

syssi, 2023-06-25:
I've increased the `rx_timeout` of the triple charger configuration example.

syssi, 2023-06-25:
Feel free to create new issues if you miss something or if there is room for improvements!

syssi, 2025-03-03:
This PR (#59) will resolve most of the mentioned issues!

PR #44 (2023-06-21): Improve triple charger support
Closes: #43