syssi/esphome-votronic #46: What I'm doing wrong?

avfoto, 2024-01-14:
I'm trying to connect with my Voltronic Axpert VM III 3000/24 with a Esphome device.
I get this error: "INFO ESPHome 2023.12.5
INFO Reading configuration /config/esphome/esphome-web-a65af0.yaml...
INFO Cloning https://github.com/syssi/esphome-votronic.git@drop-patched-ble-stack
Failed config

external_components: [source <unicode string>:20]
  - 
    Remote branch drop-patched-ble-stack not found in upstream origin.
    source: github://syssi/esphome-votronic@drop-patched-ble-stack
    refresh: 0s"

This is the code I'm using:

substitutions:
  name: esphome_voltronic
  device_description: "Monitor a votronic device via BLE"
  external_components_source: github://syssi/esphome-votronic@drop-patched-ble-stack
  mac_address: MAC_OF_MY_VOLTRONIC

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-votronic"
    version: 3.0.0

esp32:
  board: wemos_d1_mini32
  framework:
    type: esp-idf

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: "myssid"
  password: "mypass"

ota:

logger:
  level: DEBUG

# If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
# The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
mqtt:
  broker: BROKER_IP
  username: admin
  password: admin
  id: esp_voltronic

# api:

esp32_ble:
  io_capability: keyboard_only

esp32_ble_tracker:
  on_ble_advertise:
    then:
      - lambda: |-
          if (x.get_name().rfind("votronic", 0) == 0) {
            ESP_LOGI("ble_adv", "New Votronic BLE controller found");
            ESP_LOGI("ble_adv", "  Name: %s", x.get_name().c_str());
            ESP_LOGI("ble_adv", "  MAC address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
              ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
          }

ble_client:
  - mac_address: ${mac_address}
    id: client0
    on_passkey_request:
      then:
        - ble_client.passkey_reply:
            id: client0
            passkey: 173928

votronic_ble:
  - ble_client_id: client0
    id: votronic0
    throttle: 5s

binary_sensor:
  - platform: votronic_ble
    votronic_ble_id: votronic0
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
    controller_active:
      name: "${name} controller active"
    current_reduction:
      name: "${name} current reduction"
    aes_active:
      name: "${name} aes active"

sensor:
  - platform: votronic_ble
    votronic_ble_id: votronic0
    # Battery computer
    battery_voltage:
      name: "${name} battery voltage"
    secondary_battery_voltage:
      name: "${name} secondary battery voltage"
    battery_capacity_remaining:
      name: "${name} battery capacity remaining"
    state_of_charge:
      name: "${name} state of charge"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    battery_nominal_capacity:
      name: "${name} battery nominal capacity"

    # Solar charger
    pv_voltage:
      name: "${name} pv voltage"
    pv_current:
      name: "${name} pv current"
    battery_status_bitmask:
      name: "${name} battery status bitmask"
    pv_controller_status_bitmask:
      name: "${name} pv controller status bitmask"
    charged_capacity:
      name: "${name} charged capacity"
    charged_energy:
      name: "${name} charged energy"
    pv_power:
      name: "${name} pv power"

text_sensor:
  - platform: votronic_ble
    votronic_ble_id: votronic0
    battery_status:
      name: "${name} battery status"
    pv_controller_status:
      name: "${name} pv controller status"

switch:
  - platform: ble_client
    ble_client_id: client0
    name: "${name} enable bluetooth connection"

syssi, 2024-01-14:
You are in the wrong track! :-) This project is able to talk to devices of the german manufacturer called "votronic". Your are looking for a project supporting your Voltronic inverter. In best case your inverter is a PIP compatible device and the `pipsolar` component is what you are looking for. What do you think?

avfoto, 2024-01-14:
Sorry for my mistake !!

closed 2024-01-14 by avfoto