syssi/esphome-votronic #48: Invalid Header 0xFF

bavariancarp, 2024-04-28:
Hi,

I tried actually for 3 days to get your code running. Actually in MQTT is coming the message: invalid header 0xFF.

Here is my yaml:

substitutions:
  name: votronic-solar-charger
  device_description: "Monitor a Votronic Solar Charger via the display link port (UART)"
  external_components_source: github://syssi/esphome-votronic@main
  tx_pin: GPIO4
  rx_pin: GPIO5
  rx_timeout: 150ms

esphome:
  name: ${name}
  comment: ${device_description}
  project:
    name: "syssi.esphome-votronic"
    version: 3.0.0

esp8266:
  board: d1_mini

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  manual_ip: 
    static_ip: 192.168.8.108
    gateway: 192.168.8.100
    subnet: 0.0.0.0
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
logger:

# If you use Home Assistant please remove this `mqtt` section and uncomment the native `api` component!
# api:

mqtt:
  broker: 192.168.8.100
  port: 1883
  username: mqtt-benutzer
  password: anfangende123
  client_id: smartshunt

uart:
  - id: uart_modbus
    baud_rate: 1000
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}

votronic:
  - id: votronic0
    uart_id: uart_modbus
    rx_timeout: ${rx_timeout}
    throttle: 2s

binary_sensor:
  - platform: votronic
    votronic_id: votronic0
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"

sensor:
  - platform: votronic
    votronic_id: votronic0
    battery_voltage:
      name: "${name} battery voltage"
    secondary_battery_voltage:
      name: "${name} secondary battery voltage"
    battery_capacity_remaining:
      name: "${name} battery capacity remaining"
    current:
      name: "${name} current"
    power:
      name: "${name} power"
    battery_nominal_capacity:
      name: "${name} battery nominal capacity"
    state_of_charge:
      name: "${name} state of charge"
    charging_mode_setting_id:
      name: "${name} charging mode setting id"
    battery_status_bitmask:
      name: "${name} battery status bitmask"

text_sensor:
  - platform: votronic
    votronic_id: votronic0
    charging_mode_setting:
      name: "${name} charging mode setting"
    battery_status:
      name: "${name} battery status"

I hope someone can help me, to get it running.

Kind regards,

Mario

syssi, 2024-04-29:
Could you provide a photo of your setup? I would like to make sure the ESP is connected properly to the votronic device.

bavariancarp, 2024-04-29:
of course I can. the logic converter is needed, before I didn't have any incoming signal. and excuse the same color of the wires. 
[image]

syssi, 2024-04-29:
Could you give it another try to upload the image? Thanks in advance!

Receiving `0xFF` is most of the time a grounding issue and just garbage. ;-) Please don't expect the logic converter is doing a good job at the moment.

bavariancarp, 2024-04-29:
[image]

bavariancarp, 2024-05-01:
Now I have some entities working. Panel voltage, volt current, and the other panel and loading stats. but I am still missing load state and the other text sensors

syssi, 2024-05-02:
What did you change? Which Votronic devices do you own?

bavariancarp, 2024-05-02:
Puh, I have to destroy my dream, that it's working...
I set up my VICTRON that is working now, not the VOTRONIC. 😵‍💫 
I use the votronic smart shunt, that is still not working. but I checked every cable and do have still the same problem. I also changed the uart ports, checking if that is changing something. Still getting the 0xFF failure

bavariancarp, 2024-05-02:
Ich habe dir mal einen Log angehängt, eventuell bringt uns der weiter. Ist allerdings ein MQTT output
[logs_esp8266-smart-shunt-example_run.txt](https://github.com/syssi/esphome-votronic/files/15193101/logs_esp8266-smart-shunt-example_run.txt)

syssi, 2024-05-02:
Kannst du noch zusätzlich den `debug`-Modus der `uart`-Komponente aktivieren und ein weiteres Log zur Verfuegung stellen? Danke!

```
uart:
  - id: uart_0
    baud_rate: 1000
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}
    debug:
      direction: BOTH
      dummy_receiver: false
```

bavariancarp, 2024-05-02:
kein Problem:

[Uploading logs_esp8266-smart-shunt-example_run-2.txt…]()

syssi, 2024-05-02:
Irgendwas ist leider wieder beim Upload schief gelaufen. Kannst du es noch einmal versuchen? :-)

bavariancarp, 2024-05-02:
[logs_esp8266-smart-shunt-example_run-2.txt](https://github.com/syssi/esphome-votronic/files/15194196/logs_esp8266-smart-shunt-example_run-2.txt)

syssi, 2024-05-03:
Ein Votronic Display Link Frame beginnt immer mit dem Wert `0xAA` und hat eine feste Länge von 16 Bytes. Wären in deinem Logbuch valide Daten enthalten, müsste man alle 16 Bytes den Wert `0xAA` finden. Im besten Fall würde man sogar eine kurze Pause vor diesem Byte sehen. Stattdessen findet man jedoch nur Datenmüll:

```
[22:28:01][D][uart_debug:114]: <<< 7C:7F:91:A6:EF:63:39:59:ED:DD:09:EB:59:DD:77:0D:97:B5:73:4E:B0:2E:76:0F:2C:D4:B8:A5:2A:3D:F8:EB:3D:91:6F:75:B7:A9:AA:A7:DB:FB:EC:3A:5F:46:6D:F9:7D:91:E1:FC:59:9F:ED:F2:F7:AF:64:98:80:B1:77:80:DF:A8:3F:5E:45:72:FE:9E:4B:7F:7D:FF:CF:4F:2F:77:37:FB:3E
[22:28:02][D][uart_debug:114]: <<< B8:DE:91:55:18:26:C4:65:46:58:41:BA:4D:EB:39:98:56:FA:4D:FD:FB:FF:F6:BE:FE:E6:BE:FF
[22:28:07][D][uart_debug:114]: <<< A7:7B:3D:FF
[22:28:20][D][uart_debug:114]: <<< 34:E1:FE:CC:CF:5E:10:37:FC:1B:9B:ED:42:6C:1F:B8:EF:BE:90:8B:77:56:16:98:6E:B2:5D:EA:FF:DB:9D:77:8E:74:E4:D8:D1:EF:25:89:FE:DE:4D:F5:59:7D:BF:FC:89:6E:3F:C0:BC:38:DA:E5:16:ED:50:15:CA:FB:C0:42:55:8A:5B:7E:D7:8D:FC:5E:B3:21:00:0A:95:FB:FF:D5:F9:BB:BA:DF:DF:6E:F6:F6:86:F5:5E:52:09:92:FF:0A:6B:B3:13:DA:9F:49:F3:F7:FA:D0:19:0A:A3:CC:87:72:FD:33:3E:EC:BE:DB:DA:7F:A6:FF:B9:37:E7:05:BB:75:F6:3F:48:DF:EF:EC:F7:7D:EF:3C:6B:74:EC:DB:BF:D3:CD:C0:76:FD:8A:47
[22:28:24][D][uart_debug:114]: <<< 74:EC:FF:D7:B7:CC:DF:F3:AF:7B:FE:F9:93:AA:C9:9C:FA:30:FD:52:BD:CF:6E:17:3A:D4:4D:A7:09:F6:93:1C:B2:70:24:FC:6E:FA:7F:46:91:CF:B7:4B:4F:34:1A:27:13:87:AB:77:C7:B6:8B:DF:DF:A4:4E:4A:2A:DA:BC:AC:2A:CB:0A:5D:E4:1C:B8:8D:F3:4B:5B:39:DA:D8:3B:D6:6B:CF:FE:6E:5F:7D:79:FD:F9:27:A2:DB:D2:D6:69:8D:F6:FF:E7:6C:09:FA:FD:9B:AE:63:C9:D3:9F:F2:D5:64:B5:5A:3C:5A:8B:EB:C3:FC:63:FE:C6:BF:67:ED:0E:5E:30:E8:BF:92:1E:9F:3E:7C:FB:85:6F:FC:38:FF:26:EF:F6:65:F9:E6:C7:FE
[22:28:29][D][uart_debug:114]: <<< 14:AD:BB:59:CB:E2:97:6C:3A:0F:A1:C1:DA:C8:7A:59:6B:FB:95:3D:6E:D9:35:BE:D3:FF:FB:CF:D2:FF
[22:28:36][D][uart_debug:114]: <<< 0F:D5:16:DA:FF:FF:72:AA:6D:88:56:2F:BE:74:ED:FF:FC:7C:DF:FF:6F:5F:FB:69:27:CA:EC:F0:D7:37:E1:11:19:E2:4A:B0:95:3D:85:EC:65:6D:D3:FD:16:90:A9:47:80:FD:59:0D:8C:F0:19:C1:E7:5A:9D:FC:BF:E6:76:E7:E9:89:CC:DB:24:F5:49:92:83:F9:AB:A1:BE:84:3A:C6:E9:7C:7D:AC:7E:27:43:F5:F8:9D:97:CC:1D:4C:D7:1B:F5:CF:3E:E1:FE:EB:DB:B2:05:D8:60:7E:A7:07:DA:36:3E:E7:4E:9A:BC:84:68:7F:E1:CE:8C:9C:E5:D3:B9:A6:FF:F7:53:E7:B1:FB:33:33:DE:E5:F6:3E:12:BA:7B:B8:83:E5:C6:15:C8:7C
[22:28:36][D][uart_debug:114]: <<< F3:F2:A2:58:C5:CF:93:4E:AD:FF:FA:5E:F9:EA:FD:27:B5:79:67:BF:6A:A0:FA:19:5D:B9:F7:B6:FF:BB:FF:F7:4E:B7:7F:FF:FC
[22:28:41][D][uart_debug:114]: <<< F8:63:47:F0:02:A9:9E:FD:23:01:B5:7E:26:06:88:F0:DF:7B:FE:FF:AB:3E:FF:D5:DF:EF
[22:28:54][D][uart_debug:114]: <<< 4F:97:5E:DF:50:0C:6E:AE:3A:AF:97:57:78:F3:E4:8A:7E:9D:3F:E3:EF:3A:AA:1B:71:3D:F3:21:20:EA:D1:F1:71:41:53:55:A5:95:BD:D0:53:FD:D1:96:13:C7:C7:1F:F0:27:51:37:FA:36:0B:7C:B0:BD:F7:EF:FE:DF:77:FE:FF:CF:EF:FF:AF:05:94:5C:06:C6:FB:8B:58:67:BF:EA:1D:81:0D:EB:75:EC:38:78:DE:73:68:83:0A:E6:CA:D7:97:F0:F2:E4:FB:9B:AF:FB:FF:73:FF:FE:7E:4A:F3:39:A6:F5:E2:21:EF:0F:F7:45:C0:FA:8E:7E:3A:76:ED:E7:85:D9:83:69:AE:DD:FF:B1:6D:F8:C7:87:FB:06:5E:FB:EB:B7:EE:76:F7:6F
[22:28:54][D][uart_debug:114]: <<< B9:A2:98:C3:FD:34:FE:E9:7E:E6:BA:FB:3D:DF:1F:DD:A2:89:AF:7F:B1:ED:C5:C3:6A:D2:32:77:70:5A:DC:EE:F6:2B:52:EC:18:FA:7E:FF:43:16:69:69:DC:F4:6B:F2:40:01:1A:9C:39:8F:9D:8F:5B:DF:7D:D7:3B:7F:8A
[22:28:55][D][uart_debug:114]: <<< 20:F7:18:35:AF:06:FD:97:5F:5B:96:FC:EF:BD:BB:58:FF:77:FD
[22:29:03][D][uart_debug:114]: <<< 9E:EF:F6:F6:DB:F2:17:EF:E9:FE:F7:FE:01:38:BD:A4:F5:D4:D5:CA:98:C3:99:D3:05:3D:23:6A:F2:5F:45:CF:30:AE:E7:42:F3:E1:71:3C:57:77:33:E9:1B:8E:D1:D6:76:97:86:32:B7:FB:17:D7:AD:F9:A6:F7:FA:5B:08:4D:B6:E0:47:26:67:EB:E5:37:23:66:5D:2D:0B:CA:00:E9:2A:9A:F2:B4:3D:F5:7A:F1:3A:B8:FD:B8:E0:8D:D2:3E:8F:DA:6E:B1:77:AE:C6:FB:EF:EE:69:FB:FF:CF:BF:B9:77:FF:3D:DF:E6:D4:67:DA:8D:F6:FD:C5:7B:07:D3:EF:24:2A:AA:F9:CB:FF:7F:6F:4D:70:76:75:5A:99:CD:3A:72:EC:BB:63:61:9B
```

Kannst du einmal die Spannung zwischen GND und TX des Votronic-Gerätes messen? Ich würde gerne für die nächsten Experimente den Logik-Level-Konverter weglassen und möchte sicher gehen, dass weniger als 5V anliegen, wenn das Gerät sendet.

bavariancarp, 2024-05-23:
sorry für die späte Antwort. Aber ich hab nun alles zerlegt und step by step wieder zusammen gelötet, ohne irgendwelche Änderungen. Jetzt geht es und läuft stabil seit einer Woche. Vielen Dank für die Hilfe!

closed 2024-05-23 by syssi